tuna-f1sh/cyme #2: Thread 'main' panicked at 'not yet implemented: Unsupported speed'

pro-sumer, 2023-01-17:
cyme 1.2.4 fails on my (Intel) Mac. Full stack trace (`RUST_BACKTRACE=full`):

```
thread 'main' panicked at 'not yet implemented: Unsupported speed', src/usb.rs:406:22
stack backtrace:
   0:        0x106ed3100 - __mh_execute_header
   1:        0x106e21a5a - __mh_execute_header
   2:        0x106eaa43c - __mh_execute_header
   3:        0x106ed6663 - __mh_execute_header
   4:        0x106ed642a - __mh_execute_header
   5:        0x106ed6b34 - __mh_execute_header
   6:        0x106ed6814 - __mh_execute_header
   7:        0x106ed6778 - __mh_execute_header
   8:        0x106ed6742 - __mh_execute_header
   9:        0x106ee17b3 - __mh_execute_header
  10:        0x106e42c8b - __mh_execute_header
  11:        0x106e428b0 - __mh_execute_header
  12:        0x106e425a3 - __mh_execute_header
  13:        0x106e4d30d - __mh_execute_header
  14:        0x106e6530a - __mh_execute_header
  15:        0x106db83d5 - __mh_execute_header
  16:        0x106d90f66 - __mh_execute_header
  17:        0x106dc9400 - __mh_execute_header
zsh: abort      cyme
```

tuna-f1sh, 2023-01-25:
My mistake, I missed the low speed match for display. I'm 98% it's fixed but can you test 1.2.5: https://github.com/tuna-f1sh/cyme/releases/tag/v1.2.5

If you're using the brew tap, that is updated so you can just `brew update`, `brew upgrade`.

pro-sumer, 2023-01-25:
The crash happened when I had specific hardware connected.

I don't have access to that hardware until sometime next week, but I have put a task on my todo list to test this then, so I hope to have feedback next week.

tuna-f1sh, 2023-01-26:
Ok I'll close it for now. Please re-open if you have the issue again with the low speed hardware.

commit b274e7e5 (2023-01-25): fix #2; missing match for LowSpeed
closed 2023-01-26 by tuna-f1sh