tuna-f1sh/cyme #3: cyme.json not read

Mellbourn, 2023-01-20:
I can't for the life of me make cyme find the config file without a parameter.

Using the `--config=` parameter I can force cyme to use my `cyme.json` file. But without the parameter, cyme does not find the file.

I'm on macOS and I've tried these paths:

* `$HOME/.config/cyme/cyme.json`
* `$HOME/Library/Application Support/cyme/cyme.json`
* `$HOME/.config/cyme.json`
* `$HOME/Library/Application Support/cyme.json`

I don't know Rust, but in the code, it looks to me like [the CONF_NAME variable](https://github.com/tuna-f1sh/cyme/blob/d9812f1591e17a091d098091da512b14ddbb7a5b/src/config.rs#L14) isn't actually used.

tuna-f1sh, 2023-01-25:
Hmm, I just double checked and with '$HOME/Library/Application Support/cyme/cyme.json' it loads the config file there.

[CONF_NAME is used when looking](https://github.com/tuna-f1sh/cyme/blob/main/src/config.rs#L73). Can you run with the debug argument `-Z`: `cyme -Z`. It should show something like:

```
2023-01-25T14:23:32.557Z INFO  [cyme::config] Looking for cyme system config "/Users/john/Library/Application Support/cyme/cyme.json"
2023-01-25T14:23:32.557Z INFO  [cyme::config] Loaded cyme system config Config { icons: IconTheme { user: None, tree: None }, colours: ColourTheme { name: Some(BrightBlue), serial: Some(Green), manufacturer: Some(Blue), driver: Some(Cyan), string: Some(Blue), icon: None, location: Some(Magenta), path: Some(Cyan), number: Some(Cyan), speed: Some(Magenta), vid: Some(BrightYellow), pid: Some(Yellow), class_code: Some(BrightYellow), sub_code: Some(Yellow), protocol: Some(Yellow), attributes: Some(Magenta), power: Some(Red), tree: Some(BrightBlack), tree_bus_start: Some(BrightBlack), tree_bus_terminator: Some(BrightBlack), tree_configuration_terminator: Some(BrightBlack), tree_interface_terminator: Some(BrightBlack), tree_endpoint_in: Some(Yellow), tree_endpoint_out: Some(Magenta) }, blocks: None, bus_blocks: None, config_blocks: None, interface_blocks: None, endpoint_blocks: None, mask_serials: None, lsusb: false, tree: true, verbose: 0, more: false, hide_buses: true, hide_hubs: true, decimal: false, no_padding: false, ascii: false, headings: false, force_libusb: false }
```

Thanks.

Mellbourn, 2023-01-25:
Ok, that flag shows that there is a problem with my config file.
```
cyme -z
2023-01-25T15:07:06.347Z INFO  [cyme::config] Looking for cyme system config "/Users/klas.mellbourn/Library/Application Support/cyme/cyme.json"
2023-01-25T15:07:06.348Z WARN  [cyme::config] Failed to read cyme system config "/Users/klas.mellbourn/Library/Application Support/cyme/cyme.json": Error(missing field `lsusb` at line 24 column 1)
2023-01-25T15:07:06.470Z INFO  [cyme] Filtering with None
```

This is the file
```json
{
  "blocks": ["bus-number", "device-number", "icon", "name", "speed"],
  "bus-blocks": ["name", "host-controller"],
  "config-blocks": ["number", "name", "icon-attributes", "max-power"],
  "interface-blocks": [
    "port-path",
    "icon",
    "alt-setting",
    "class-code",
    "sub-class",
    "protocol",
    "name"
  ],
  "endpoint-blocks": [
    "number",
    "direction",
    "transfer-type",
    "sync-type",
    "usage-type",
    "max-packet-size"
  ],
  "tree": true,
  "headings": false
}
```

Mellbourn, 2023-01-25:
I got this config file working
```json
{
  "blocks": ["bus-number", "device-number", "icon", "name", "speed"],
  "bus-blocks": ["name", "host-controller"],
  "config-blocks": ["number", "name", "icon-attributes", "max-power"],
  "interface-blocks": [
    "port-path",
    "icon",
    "alt-setting",
    "class-code",
    "sub-class",
    "protocol",
    "name"
  ],
  "endpoint-blocks": [
    "number",
    "direction",
    "transfer-type",
    "sync-type",
    "usage-type",
    "max-packet-size"
  ],
  "tree": true,
  "lsusb": false,
  "verbose": 0,
  "more": false,
  "hide-buses": false,
  "hide-hubs": false,
  "decimal": false,
  "no-padding": false,
  "ascii": false,
  "headings": false,
  "force-libusb": false
}
```

So, it did find the correct file. But maybe errors that occur when reading the config file should be reported?

tuna-f1sh, 2023-01-25:
Yes, I've added a fix and test for this now - I missed the macro to de-serialise with defaults for all fields. Thanks for helping to debug.

commit cc3c1f19 (2023-01-25): fix #3; deserialize with default args for all config fields
closed 2023-01-25 by tuna-f1sh