tuna-f1sh/cyme #4: Abort trap: 6

joedupa, 2023-01-22:
what am I doing wrong? this is on Maverick 10.14 OS:

Gs-MBP-7:~ g$ cyme
dyld: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
Referenced from: /usr/local/bin/cyme
Reason: image not found
Abort trap: 6
Gs-MBP-7:~ g$

Thanks!!

tuna-f1sh, 2023-01-25:
Looks like it is trying to call libusb but you don't have it. 

I originally didn't require libusb but thinking about it, the pre-compiled release will compile with it and so require it. You can either install with `brew install libusb` for you'll have to compile cyme from source with `cargo install --no-default-features cyme`

tuna-f1sh, 2023-01-25:
I've also updated the homebrew tap so that libusb is a 'depends_on'.

joedupa, 2023-01-26:
now there is another error:

Gs-MBP-7:~ g$ Gs-MBP-7:~ g$ cyme -l -z
2023-01-26T13:02:52.027Z INFO  [cyme::config] Looking for cyme system config "/Users/g/Library/Application Support/cyme/cyme.json"
Failed to parse system_profiler output: Error(EOF while parsing a value at line 1 column 0)

I reinstalled cyme with brew and this file is missing.

tuna-f1sh, 2023-01-26:
This is actually not the cyme.json file missing; it looked but found nothing so did not attempt to read - it due to it failing to parse the output of `system_profiler -json SPUSBDataType`.

Can you run that command above and share the output? I can debug it then.

tuna-f1sh, 2023-01-31:
I pushed an update (1.2.7), that will fallback to libusb if `system_profiler` returns non-zero. I think this is the problem, not a parsing error here. I expect that 10.14 doesn't include the `-json` flag.

closed 2023-03-08 by tuna-f1sh