tuna-f1sh/cyme #5: compile error: undefined reference to `getauxval'

bAndie91, 2023-01-26:
it does not find getauxval symbol.
i gave it the path to libc6 2.29.
what libs or .a archives should i give it?

```
Compiling cyme v1.2.4 (/.../cyme)
error: linking with `cc` failed: exit status: 1
...
  = note: /.../cyme/target/release/deps/cyme-bb3b6e2621ca5f78.cyme.5bea5bcc-cgu.0.rcgu.o: In function `rustix::backend::vdso_wrappers::init_syscall::h31c0d04ab5f7089d':
          cyme.5bea5bcc-cgu.0:(.text._ZN6rustix7backend13vdso_wrappers12init_syscall17h31c0d04ab5f7089dE+0x3b): undefined reference to `getauxval'
          collect2: error: ld returned 1 exit status
...
```

tuna-f1sh, 2023-01-26:
What toolchain are you using? The stable release? I don't think this is a problem with cyme but a issue with your toolchain/Rust: https://github.com/rust-lang/rust/issues/89626

bAndie91, 2023-01-27:
it's `+nightly-i686-unknown-linux-gnu`.
the toolchain should be also OK, because on an other distro, it compiles.
i have libc6 on a non standard path, that must be the issue.

this is the build cmd, btw:
`PATH=$HOME/.cargo/bin:/usr/local/opt/gcc_8.5.0/bin:$PATH  PKG_CONFIG_PATH=/usr/local/opt/libusb_1.0.19/lib/pkgconfig RUSTFLAGS=-L/usr/local/opt/libc6/2.28/lib cargo +nightly-i686-unknown-linux-gnu build --release`

tuna-f1sh, 2023-01-27:
Ok well I'm closing since it's not to do with cyme and something to do with your environment as you say.

closed 2023-01-27 by tuna-f1sh