tuna-f1sh/cyme #7: cyme on Apple silicon

applebait, 2023-02-17:
Probably I need to manage how to link libusb to where cyme is looking for it…
I'm getting this output when trying "cyme" or "cyme --force-libusb":

dyld[20594]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <5D31FBC5-C545-38BC-A8C4-49BD63503AA7> /opt/homebrew/Cellar/cyme/1.2.7/bin/cyme
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
zsh: abort      cyme --force-libusb

tuna-f1sh, 2023-02-20:
Hmm, it looks like this is due to libusb not being installed (`brew install libusb`) I added it as a [depends_on](https://github.com/tuna-f1sh/homebrew-taps/blob/main/Formula/cyme.rb#L8) so not sure what's going on. 

If you do, `brew info cyme` do you get:

```
==> Dependencies
Required: libusb ✔
```

As part of the output?

tuna-f1sh, 2023-02-20:
I think this is as you say, due to libusb not being where expected 

```
export DYLD_PRINT_LIBRARIES=1                                                                                                                                 
cyme                                                                                                                                                          
dyld[54812]: <5D31FBC5-C545-38BC-A8C4-49BD63503AA7> /usr/local/Cellar/cyme/1.2.7/bin/cyme                                                                       
dyld[54812]: <3B6C447C-8F51-36D5-9717-27C8AAC5C0ED> /usr/local/Cellar/libusb/1.0.26/lib/libusb-1.0.0.dylib
```

Is there anything in `ls -la /usr/local/opt/libusb/lib` ? There should be a symbolic link created by Homebrew:

```
ls -la /usr/local/opt/ | grep 'libusb'
lrwxr-xr-x john admin  23 B  Mon Apr 25 16:16:57 2022 libusb ⇒ ../Cellar/libusb/1.0.26
```

webhdx, 2023-02-20:
Hello. I ran into the same issue on my Apple Studio M1 Max. I have `libusb` installed:

```
➜ brew info cyme
==> tuna-f1sh/taps/cyme: stable 1.2.7
List system USB buses and devices; a modern and compatiable `lsusb`
https://github.com/tuna-f1sh/cyme
/opt/homebrew/Cellar/cyme/1.2.7 (9 files, 3.1MB) *
  Built from source on 2023-02-20 at 17:56:42
From: https://github.com/tuna-f1sh/homebrew-taps/blob/HEAD/Formula/cyme.rb
License: GPL-3.0-or-later
==> Dependencies
Required: libusb ✔
==> Caveats
zsh completions have been installed to:
  /opt/homebrew/share/zsh/site-functions
```

and get the same error when I run `cyme`:

```
➜ cyme
dyld[5391]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <5D31FBC5-C545-38BC-A8C4-49BD63503AA7> /opt/homebrew/Cellar/cyme/1.2.7/bin/cyme
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
zsh: abort      cyme
```

Looks like `libusb` is not properly linked:
```
➜ ls -la /usr/local/opt/libusb/lib
ls: /usr/local/opt/libusb/lib: No such file or directory
```

I also tried running `brew unlink libusb && brew link libusb` but it doesn't help. The `*.dylib` file is present in `/opt/homebrew/Cellar/libusb/1.0.26/lib`:

```
➜ ls -la /opt/homebrew/Cellar/libusb/1.0.26/lib | grep libusb
-r--r--r--@  1 maciejkobus  admin  177440 19 lut 00:53 libusb-1.0.0.dylib
-r--r--r--@  1 maciejkobus  admin  194408 10 kwi  2022 libusb-1.0.a
lrwxr-xr-x@  1 maciejkobus  admin      18 10 kwi  2022 libusb-1.0.dylib -> libusb-1.0.0.dylib
```

tuna-f1sh, 2023-02-20:
Thanks for the extra info. It looks like ARM binaries and libs are stored in a different place: https://apple.stackexchange.com/a/410829

I'll have to look into it a bit more to figure out what's going on and how tools account for both. It might be that the build workflow needs to create a binary for macOS x86 and ARM. Any extra digging by you guys would be helpful since I don't have a Apple silicon to test with.

webhdx, 2023-02-20:
If you have anything to test just ping me. I love the project and since I mess with USB devices a lot I'd greatly appreciate being able to use it on Apple Silicon :)

webhdx, 2023-02-20:
BTW I can confirm `libusb-1.0.0.dylib` is present in `/opt/homebrew/lib/` on my Mac.

tuna-f1sh, 2023-02-20:
Can you try this cross-compiled aarch64? [cyme-aarch64-apple-darwin.zip](https://github.com/tuna-f1sh/cyme/files/10786942/cyme-aarch64-apple-darwin.zip) - I'll look at adding a universal binary if it does.

It might also work if you `export DYLD_LIBRARY_PATH=/opt/homebrew/lib` before running the current release.

webhdx, 2023-02-20:
Yup precompiled binary works. Great job. Having native aarch64 binary will be preferred over x86 Rosetta translated one. Hope this is added to homebrew soon.

closed 2023-02-21 by tuna-f1sh

applebait, 2023-02-21:
Sorry for being out of communication… And a million thanks to both of you for managing the fix!

commit ecae51c9 (2023-02-21): fix #7; workflow builds universal binary
commit 1a4ef221 (2023-02-21): fix #7; workflow builds universal binary