tuna-f1sh/cyme #8: Truncate very long fields

dmezh, 2023-02-20:
Hey, amazing work on this; I like it a lot! I suggest truncating very long names - motivating example:

[image]

That one device with the extremely long name is pushing the serial column off to the side and hurting readability. I would prefer the name ends in a `...` after some default max, and a flag to never truncate.

tuna-f1sh, 2023-02-21:
Yes this is a good idea. It's actually something I have in my list of features to implement - some terminal width detection and auto-scaling. But as you say, a fixed max also makes sense for descriptor strings.

I'll have a look into it next time I get chance.

tuna-f1sh, 2023-02-27:
I had added this and whilst I was at it, added a default auto-width feature that will find the max based on terminal width. See: https://github.com/tuna-f1sh/cyme/releases/tag/v1.3.1

closed 2023-02-27 by tuna-f1sh