tuna-f1sh/cyme #11: Invalid character in the `I` column

phuclv90, 2023-07-05:
When I tried cyme in both the default macOS terminal or iTerm2 I always get some private U+F287 characters that's undisplayable. Why is that and is there anyway to fix it?

[image]

tuna-f1sh, 2023-07-05:
It's most probably because the font are using is missing glyphs. I realise I haven't added a note on this to my repository but the one in `lsd` should be applicable for you to sort this: https://github.com/lsd-rs/lsd#icons-not-showing-up

tuna-f1sh, 2023-07-05:
I've updated the README: https://github.com/tuna-f1sh/cyme#custom-icons-and-colours

Note that you can either install a nerd font, remove the icon column with a custom config or use only `--ascii` chars.