tuna-f1sh/cyme #12: Using cyme as a library

haata, 2023-10-02:
First of all, great tool! I've always found the lack of USB info on Windows and macOS especially annoying.

This is more of a request. I have a gui application that assists with device flashing and QC steps during manufacturing.
One of the things that this application can do is print out the current `lsusb` in a small window (`dfu-util -l` as well) to help debug a newly inserted USB device when it's not detected.

It would be very useful to be able to call cyme as a library rather than an external binary, so the tool is self contained (i.e. when I need to run it on a Windows system, or reducing the number of external dependencies).

I've briefly looked at the source for cyme and I'm not really sure how easy to do this would be (as I need string output, preferably pretty printed, one line per device).

tuna-f1sh, 2023-10-03:
Something like this?

```
use cyme::lsusb::profiler;

fn main() -> Result<(), String> {
    // get all system devices - use get_spusb_with_extra for verbose info
    let mut sp_usb = profiler::get_spusb(false).map_err(|e| {
        format!("Failed to gather system USB data from libusb, Error({})", e)
    })?;

    // flatten since we don't care tree/buses
    let devices = sp_usb.flatten_devices();

    for device in devices {
        println!("{}", device.to_lsusb_string());
    }

    Ok(())
}
```
This will print in the same way that `lsusb` prints the list of devices. If you want to pretty print like `cyme` see the 'display' module's print functions: https://docs.rs/cyme/1.5.0/cyme/display/index.html. For example, with that flattened devices, one can print with [PrintSettings](https://docs.rs/cyme/1.5.0/cyme/display/struct.PrintSettings.html):

```
use cyme::lsusb::profiler;
use cyme::display;

fn main() -> Result<(), String> {
    // get all system devices - use get_spusb_with_extra for verbose info
    let mut sp_usb = profiler::get_spusb(false).map_err(|e| {
        format!("Failed to gather system USB data from libusb, Error({})", e)
    })?;

    // flatten since we don't care tree/buses
    let devices = sp_usb.flatten_devices();

    cyme::display::print_flattened_devices(&devices, &display::PrintSettings::default());

    Ok(())
}
```

haata, 2023-10-03:
THANK YOU!

(it might be useful to add this to the docs or as an example somewhere)

commit 78eee26c (2023-10-03): fix #12 basic lib example; lib.rs doc example
closed 2023-10-03 by tuna-f1sh