tuna-f1sh/cyme #17: (Question) Do you plan to rewrite libusb in Rust?

hardBSDk, 2024-05-09:


tuna-f1sh, 2024-05-11:
Not personally but [nusb](https://github.com/kevinmehall/nusb) is on my radar. It would be a pure Rust solution rather than libusb but not explored whether porting would be possible yet. Open to contributions if you're interested in exploring.

tuna-f1sh, 2024-08-16:
@hardBSDk if you interested in testing this, the PR #26 can be built with pure Rust using `--no-default-features -F=native`.

hardBSDk, 2024-08-17:
@tuna-f1sh It works but don't show the xHCI ports.

tuna-f1sh, 2024-08-19:
Yes this was actually a change I made because by default nusb does not list xHCI (root_hubs). It made me re-think the default list and not to show them. The lsusb mode will though (since it does): `cargo run --no-default-features -F=native -- --lsusb`. 

Do you think they are useful in the list output? I figured not since they are not something one does anything with. They are presented in `--tree` to show which root_hub devices are attached - I think this is useful.

PR #26 (2024-10-16): Replace libusb with pure Rust nusb
Creates a `Profiler` trait that provides both libusb and nusb system USB profilers. I've kept 'libusb' as an optional feature for those wishing to keep using that and for exact parity with `lsusb`; they should both output the same but `lsusb` uses libusb of course.

- [x] ~~Since nusb uses IOkit (same as `system_profiler` command), use this for all macOS profiling by default instead of parsing `system_profiler` or at least default merge if different - bus names seem to be missing.~~ left this as a TODO one day.
- [x] Refactor repeat code in the profilers so t [... cut]
closed 2024-10-16 by tuna-f1sh