tuna-f1sh/cyme #18: (Question) There's a crate to replace libudev?

hardBSDk, 2024-05-09:
Remove direct C dependencies.

tuna-f1sh, 2024-05-11:
Is the question whether there is one or a statement that there is? 

I would like to remove libudev-sys since it seems to be [unmaintained](https://github.com/dcuddeback/libudev-sys/pull/16). That pull request requires me to have a patch in Cargo.toml, which currently prevents crates.io updates.

auronandace, 2024-05-12:
If you'd like to stick to bindings then this project looks like a supported alternative: https://github.com/Smithay/udev-rs

If you'd like to try a pure rust replacement I've found this one: https://github.com/cr8t/udev

tuna-f1sh, 2024-05-13:
Thanks. udev is only required for the HWDB lookup (for 1:1 with `lsusb` since usb-ids can also be used for this) and device driver lookup. Should it should be quite easy to port to either of these. Mostly contained in these fenced functions: https://github.com/tuna-f1sh/cyme/blob/main/src/udev.rs

tuna-f1sh, 2024-05-21:
I had a look at the pure Rust [udev](https://github.com/cr8t/udev) and unfortunately it's not a crate yet. I've asked the question https://github.com/cr8t/udev/issues/20 whether it is ready for one or not.

I forgot myself that `cyme` already uses [udev-rs](https://github.com/Smithay/udev-rs) as a direct dependency and that the problematic `libudev-sys` is a sub-dependency of this. So using the pure Rust 'udev' is the only option here.

tuna-f1sh, 2024-06-13:
https://github.com/tuna-f1sh/cyme/tree/udevrs now uses udev-rs by default rather than rely on libudev. I left the option of using libudev with the `--feature=udevlib`.

I would merge it other than I have issues with hwdb lookup not passing tests (`--feature=udev_hwdb`). I've created an issue in https://github.com/cr8t/udev/issues/22 . I also want to generally test it a bit more. If @auronandace / @hardBSDk can test that would be cool.

PR #19 (2024-06-25): Replace udev-rs libudev bindings with native Rust udev 
Removes C dependency and the minimally used udev features.

Closes #18.
closed 2024-06-25 by tuna-f1sh