tuna-f1sh/cyme #20: Couple of broken symbols (Nerd Fonts)

Cocoader, 2024-06-25:
I think [Nerd Fonts](https://www.nerdfonts.com/cheat-sheet) changed a couple of symbols.
Unfortunately I don't know Rust or have the tool-chain installed to test it myself and make a PR.

https://github.com/tuna-f1sh/cyme/blob/acd9ffac95df9c2f029f52b34dc3829080a3a4f0/src/display.rs#L1637
f06a6
________________________________________________________________________________________________________________
https://github.com/tuna-f1sh/cyme/blob/acd9ffac95df9c2f029f52b34dc3829080a3a4f0/src/display.rs#L1638
f0155
________________________________________________________________________________________________________________
https://github.com/tuna-f1sh/cyme/blob/acd9ffac95df9c2f029f52b34dc3829080a3a4f0/src/icon.rs#L241
f0372
________________________________________________________________________________________________________________
https://github.com/tuna-f1sh/cyme/blob/acd9ffac95df9c2f029f52b34dc3829080a3a4f0/src/icon.rs#L248
f037d
________________________________________________________________________________________________________________
https://github.com/tuna-f1sh/cyme/blob/acd9ffac95df9c2f029f52b34dc3829080a3a4f0/src/icon.rs#L264
f0306
________________________________________________________________________________________________________________
https://github.com/tuna-f1sh/cyme/blob/acd9ffac95df9c2f029f52b34dc3829080a3a4f0/src/icon.rs#L268-L269
f0379
PS: I can't find out by the VID/PID what kind of sandisk device it is but if it is a ext. disk maybe f0a0 is better suited.
________________________________________________________________________________________________________________
https://github.com/tuna-f1sh/cyme/blob/acd9ffac95df9c2f029f52b34dc3829080a3a4f0/src/icon.rs#L276-L277
f0306
________________________________________________________________________________________________________________
https://github.com/tuna-f1sh/cyme/blob/acd9ffac95df9c2f029f52b34dc3829080a3a4f0/src/icon.rs#L279
f0841
________________________________________________________________________________________________________________
https://github.com/tuna-f1sh/cyme/blob/acd9ffac95df9c2f029f52b34dc3829080a3a4f0/src/icon.rs#L287
f030c
________________________________________________________________________________________________________________

tuna-f1sh, 2024-06-25:
Thanks! Thought I captured them all with https://github.com/tuna-f1sh/cyme/commit/f50e5f90ab572b9634ff0041e6d10a213b92294f but didn't look too hard. Fixed with https://github.com/tuna-f1sh/cyme/commit/085789de39268a6d8321dd717a687ca381c582a0

tuna-f1sh, 2024-06-25:
Re the disk - was probably one I had attached whilst developing; a SanDisk Extreme SSD I think. You're right the HDD is a bit skeuomorphic but it's clearer than the other and I like it. Idea is they are meant to be example defaults - other than common manufacturers - which one can override with a config https://github.com/tuna-f1sh/cyme?tab=readme-ov-file#custom-icons-and-colours

closed 2024-06-25 by tuna-f1sh

Cocoader, 2024-06-25:
> Thanks! Thought I captured them all with [f50e5f9](https://github.com/tuna-f1sh/cyme/commit/f50e5f90ab572b9634ff0041e6d10a213b92294f) but didn't look too hard. Fixed with [085789d](https://github.com/tuna-f1sh/cyme/commit/085789de39268a6d8321dd717a687ca381c582a0)

Absolutely no problem - I did find one or two with my setup and figured let's check all of them while I'm at it and get some inspiration for my custom icons.

> Re the disk - was probably one I had attached whilst developing; a SanDisk Extreme SSD I think. You're right the HDD is a bit skeuomorphic but it's clearer than the other and I like it. Idea is they [...]

That sounds logical. It's funny you said SanDisk Extreme SSD. As luck would have it I also have one in my setup. 
I would have guessed the PID matches but they probably have a couple of reserved PIDs for hardware / software revisions.
[image]
The two reasonable candidates I found were the hard_drive symbol (f0a0) or the usb_flash_drive symbol (f129e) - both aren't perfect but closer to reality. But considering the font size in the terminal the hard drive looks a bit more expressive.
I do like your approach with sensible defaults and a user config for customisations.

This is a great project you created.

tuna-f1sh, 2024-06-26:
Thanks for comments. Looks like it was the [Extreme Portable SSD](https://devicehunt.com/view/type/usb/vendor/0781/device/558C). Since SanDisk just makes external drives, I’m tempted to have it as a default for the vendor rather than project specific.

Cocoader, 2024-06-26:
Sounds reasonable. And SD / Micro SD cards, which SanDisk produces a lot, wouldn't be recognised by their individual VID/PID but by the VID/PID of the card reader instead.
[image]

Maybe it would be even better if you use the usb_flash_drive symbol (f129e) as default for the vendor and the hard_drive symbol (f0a0) for the Extreme Portable SSD VID/PID. SanDisk produces so many more different kinds of thumb drives and only a hand full of external SSDs. All the other (SanDisk) outliers like a card reader belong in a user config.

> Since SanDisk just makes external drives, I’m tempted to have it as a default for the vendor rather than project specific.

I see you already added this in f62d874 and pushed a new version for the previous - glad to see you respond so quickly to feedback.

Keep up the good work.😃

tuna-f1sh, 2024-06-27:
Thanks! I also added the SD Card for that card reader and one I had - it's a useful one I think. There are many however but the example should allow others to provide a config for their particular one. It made me think perhaps a device name pattern match type could be added to the `Icon` enum, for matching things like Card Readers, though it would make the icon search per device much more intensive.

If you have any other ideas for useful defaults let me know. I don't want to add too many  specific ones, since the [icon lookup](https://github.com/tuna-f1sh/cyme/blob/9667cfdfcfc208f044be81b6b9d7c6cdae795af9/src/icon.rs#L337) is run against each device but they are useful for quickly checking what is attached.

Cocoader, 2024-06-29:
> Thanks! I also added the SD Card for that card reader and one I had - it's a useful one I think. There are many however but the example should allow others to provide a config for their particular one.

I agree. 

>  It made me think perhaps a device name pattern match type could be added to the Icon enum, for matching things like Card Readers, though it would make the icon search per device much more intensive.

How laborious would be writing a test to see the performance impact of this kind of matching compared to a huge icon enum be? If this can somewhat easily be benchmarked it might turn into a future feature. But you are right - the icon lookup shouldn't impair the core functionality.

 > If you have any other ideas for useful defaults let me know. I don't want to add too many specific ones, since the icon lookup is run against each device but they are useful for quickly checking what is attached.

I will gladly do that.

tuna-f1sh, 2024-06-30:
I mocked up an implementation of the pattern match icon yesterday: https://github.com/tuna-f1sh/cyme/pull/22 - will do some performance testing and decide whether to merge it or not.