tuna-f1sh/cyme #32: thread 'main' panicked at /rustc/3f5fd8dd41153bc5fdca9427e9e05be2c767ba23/library/alloc/src/string.rs:1410:13:

undrwater, 2024-09-18:
When running 
`cyme -m`
output looks like:
```
cyme -m
  1   5 9          1-9         󰍽 0x046d 0xc08b 1b.03 2.00  0x00 UseInterfa... 0x00 -      0x00 -             G502 HERO ... Logitech      028D334B3338 usb     12.0 Mb/s
  1   4 8          1-8          0x05e3 0x0608 3c.90 2.00  0x09 Hub           0x00 Unused 0x01 Single TT     USB2.0 Hub    Genesys Lo... -            usb    480.0 Mb/s
  1   3 7          1-7          0x0b05 0x19af 1.00  2.00  0x00 UseInterfa... 0x00 -      0x00 -             AURA LED C... AsusTek Co... 9876543210   usb     12.0 Mb/s
  1   2 6          1-6          0x0b05 0x1a52 0.02  2.00  0xef Miscellaneous 0x02 ?      0x01 Interface ... USB Audio     Generic       -            usb    480.0 Mb/s
  1   6 11         1-11         0x0e8d 0x0608 1.00  2.10  0xef Miscellaneous 0x02 ?      0x01 Interface ... Wireless_D... MediaTek Inc. 000000000    usb    480.0 Mb/s
  5   2 2          5-2          0x174c 0x2074 0.01  2.10  0x09 Hub           0x00 Unused 0x02 TT per port   ASM107x       ASUS TEK.     -            usb    480.0 Mb/s
  5   4 2-4        5-2.4        0x0d8c 0x016c 1.00  1.10  0x00 UseInterfa... 0x00 -      0x00 -             USB Advanc... C-Media El... -            usb     12.0 Mb/s
  5   3 2-2        5-2.2        0x2109 0x2817 2.14  2.10  0x09 Hub           0x00 Unused 0x02 TT per port   USB2.0 Hub    VIA Labs, ... -            usb    480.0 Mb/s
  5   6 2-2-4      5-2.2.4      0x2109 0x2817 2.14  2.10  0x09 Hub           0x00 Unused 0x02 TT per port   USB2.0 Hub    VIA Labs, ... -            usb    480.0 Mb/s
  5   8 2-2-4-4    5-2.2.4.4    0x2109 0x2817 2.14  2.10  0x09 Hub           0x00 Unused 0x02 TT per port   USB2.0 Hub    VIA Labs, ... -            usb    480.0 Mb/s
  5   9 2-2-4-4-4  5-2.2.4.4.4  0x248a 0x8327 1.00  1.10  0x00 UseInterfa... 0x00 -      0x00 -             Wireless R... Telink        -            usb     12.0 Mb/s
  5   7 2-2-4-3    5-2.2.4.3   󰍽 0x046d 0xc52b c.01  2.00  0x00 UseInterfa... 0x00 -      0x00 -             USB Receiver  Logitech      -            usb     12.0 Mb/s
thread 'main' panicked at /rustc/3f5fd8dd41153bc5fdca9427e9e05be2c767ba23/library/alloc/src/string.rs:1410:13:
assertion failed: self.is_char_boundary(new_len)
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Aborted
```
dmesg shows:
`usb 1-6: usbfs: process 8504 (cyme) did not claim interface 7 before use` 
There are several of these messages with differing interface numbers.

I saw you asked for
`cyme --tree -vvv` on another issue, so here's mine:

```
cyme --tree -vvv
● xHCI Host Controller Linux 6.10.9-gentoo xhci-hcd
├──○    2 0x0b05 0x1a52 USB Audio              -           
│  └──•  1     100 mA USB Audio
│     ├──◦ 1-6:1.0  0x00 Audio 0x01 0x20 Realtek USB2.0 Audio
│     │  └──→  1 In  Interrupt None  Data  1x 16 
│     ├──◦ 1-6:1.1  0x00 Audio 0x02 0x20 Realtek USB2.0 Audio
│     ├──◦ 1-6:1.1  0x01 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  1 In  Isochronous Asynchronous Data  1x 124
│     ├──◦ 1-6:1.1  0x02 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  1 In  Isochronous Asynchronous Data  1x 186
│     ├──◦ 1-6:1.1  0x03 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  1 In  Isochronous Asynchronous Data  1x 124
│     ├──◦ 1-6:1.1  0x04 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  1 In  Isochronous Asynchronous Data  1x 186
│     ├──◦ 1-6:1.1  0x05 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  1 In  Isochronous Asynchronous Data  1x 124
│     ├──◦ 1-6:1.1  0x06 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  1 In  Isochronous Asynchronous Data  1x 186
│     ├──◦ 1-6:1.1  0x07 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  1 In  Isochronous Asynchronous Data  1x 124
│     ├──◦ 1-6:1.1  0x08 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  1 In  Isochronous Asynchronous Data  1x 186
│     ├──◦ 1-6:1.2  0x00 Audio 0x02 0x20 Realtek USB2.0 Audio
│     ├──◦ 1-6:1.2  0x01 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  3 In  Isochronous Asynchronous Data  1x 124
│     ├──◦ 1-6:1.2  0x02 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  3 In  Isochronous Asynchronous Data  1x 186
│     ├──◦ 1-6:1.2  0x03 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  3 In  Isochronous Asynchronous Data  1x 124
│     ├──◦ 1-6:1.2  0x04 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  3 In  Isochronous Asynchronous Data  1x 186
│     ├──◦ 1-6:1.2  0x05 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  3 In  Isochronous Asynchronous Data  1x 124
│     ├──◦ 1-6:1.2  0x06 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  3 In  Isochronous Asynchronous Data  1x 186
│     ├──◦ 1-6:1.2  0x07 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  3 In  Isochronous Asynchronous Data  1x 124
│     ├──◦ 1-6:1.2  0x08 Audio 0x02 0x20 Realtek USB2.0 Audio
│     │  └──→  3 In  Isochronous Asynchronous Data  1x 186
[... 216 lines cut ...]
         ├──◦ 5-2.4:1.1  0x01 Audio 0x02 0x00     
         │  └──←  1 Out Isochronous Synchronous Data  1x 200
         ├──◦ 5-2.4:1.2  0x00 Audio 0x02 0x00     
         ├──◦ 5-2.4:1.2  0x01 Audio 0x02 0x00     
         │  └──→  2 In  Isochronous Asynchronous Data  1x 200
         └──◦ 5-2.4:1.3 󰌌 0x00 HID   0x00 0x00     
            └──→  7 In  Interrupt None  Data  1x 16 

● xHCI Host Controller Linux 6.10.9-gentoo xhci-hcd
└──○    2 0x174c 0x3074 ASM107x -     
   └──•  1 󰅕 󰚥   2 mA     
      └──◦ 6-2:1.0  0x00 Hub   0x00 0x00     
         └──→  1 In  Interrupt None  Feedback 1x 2  

● xHCI Host Controller Linux 6.10.9-gentoo xhci-hcd

● xHCI Host Controller Linux 6.10.9-gentoo xhci-hcd

```
Finally:
```
cyme --version
cyme 1.8.2

```

undrwater, 2024-09-19:
dump.json attached
[dump.json](https://github.com/user-attachments/files/17066402/dump.json)

tuna-f1sh, 2024-09-20:
Thanks. I run it on my Linux system and can't get the panic so must be something live. Sorry, but can you run `RUST_BACKTRACE=1 cyme -m` to capture full backtrace and hopefully see what cyme function is causing the `is_char_boundary` panic.

tuna-f1sh, 2024-09-20:
I pushed what I think is a fix. If you can test that would be great. You can get a build from the runner if you don't have a rust toolchain installed: https://github.com/tuna-f1sh/cyme/actions/runs/10959429558

undrwater, 2024-09-20:
Yes, that works for x86_64 linux.

Thanks!

closed 2024-09-21 by tuna-f1sh