tuna-f1sh/cyme #33: lsusb compat: Devices in 'cyme -l' are sorted differently than in 'lsusb'

holgersson32644, 2024-09-21:
Hi,
your tool looks interesting, thanks for sharing it!

I noticed that the device order differs in details between `lsusb` and `cyme -l` (or `cyme --lsusb`) while you mention identical output as one of the project goals, so I wanted to provide you that feedback.

The outputs of cyme and lsusb follow here. For differences see e.g. the order of Bus1, devices 1, and 2 or Bus6 where I don’t see an ordering.

```
% cyme -l
Bus 001 Device 002: ID 13d3:5405 IMC Networks Integrated Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 058f:9540 Alcor Micro Corp. AU9540 Smartcard Reader
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 008: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 006 Device 002: ID 06cb:00bd Synaptics, Inc. Prometheus MIS Touch Fingerprint Reader
Bus 006 Device 009: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 006 Device 010: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 006 Device 011: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 006 Device 013: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 006 Device 014: ID 1ea7:0064 SHARKOON Technologies GmbH 2.4GHz Wireless rechargeable vertical mouse [More&Better]
Bus 006 Device 012: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 003: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 007 Device 004: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 007 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
% which cyme
/usr/bin/cyme
% cyme --version
cyme 1.8.2
```

```
% lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 13d3:5405 IMC Networks Integrated Camera
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 058f:9540 Alcor Micro Corp. AU9540 Smartcard Reader
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 002: ID 06cb:00bd Synaptics, Inc. Prometheus MIS Touch Fingerprint Reader
Bus 006 Device 008: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 006 Device 009: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 006 Device 010: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 006 Device 011: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 006 Device 012: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 006 Device 013: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 006 Device 014: ID 1ea7:0064 SHARKOON Technologies GmbH 2.4GHz Wireless rechargeable vertical mouse [More&Better]
Bus 007 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 003: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 007 Device 004: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
% lsusb --version
lsusb (usbutils) 017 
% which lsusb
/usr/bin/lsusb
```

I'm running on Gentoo with sys-apps/cyme-1.8.2 (build with `rustc 1.81.0 (eeb90cda1 2024-09-04) (gentoo)`), but I can reproduce this with your released binary, too.
I'm using the C locale globally in my shells, and setting LANG for both tools had no impact on the output.

tuna-f1sh, 2024-09-21:
Thanks for the report. I was actually aware that the ordering could be different but I thought it was random. I had based that on my initial development pre this change to `lsusb` where sorting was added: https://github.com/gregkh/usbutils/blob/master/lsusb.c#L3656

Now `lsusb` sorts the devices within the tree, I'll make a change to match. There is actually the `cyme --sort-devices device-number` flag but it sorts in order of device number independent of bus number so looks odd without `--tree` - need to rethink that logic. They also followed up with [this](https://github.com/gregkh/usbutils/commit/1d2524018a9c57db4801701caf41a081bee7b84a) to keep bus order so I'll do the same.

tuna-f1sh, 2024-09-23:
If you're interested in testing, I have a branch with a change that should now sort by bus number and device number on each bus. Now `--lsusb` and `lsusb` order should match. https://github.com/tuna-f1sh/cyme/pull/34

holgersson32644, 2024-09-23:
I tried it and can confirm that `cyme --lsusb` shows the ordered output now. Thanks!

PR #34 (2024-09-24): sort by bus; device-number by default and tidy logic #33
Sorting should now match updated lsusb
[1](https://github.com/gregkh/usbutils/commit/2e08ffdf19a3dc8cbb811a30d7f990d6517ea137), [2](https://github.com/gregkh/usbutils/commit/1d2524018a9c57db4801701caf41a081bee7b84a), which sorts by bus and device number within bus.

Tidied up sort logic by doing in `display::prepare` rather than at display function. Default is now `Sort::DeviceNumber` to match lsusb and I think makes most sense since device number is default display block - previously would sort on `Sort::BranchPosition` when using `--tr [... cut]
closed 2024-09-24 by tuna-f1sh