tuna-f1sh/cyme #35: Test failures for 1.8.3 on Arch Linux

orhun, 2024-09-26:
Hey, I'm getting the following test failure while building the latest version for Arch Linux (in a [clean chroot](https://wiki.archlinux.org/title/DeveloperWiki:Building_in_a_clean_chroot)):

```
failures:

---- src/lsusb/names.rs - lsusb::names::class (line 37) stdout ----
Test executable failed (exit status: 101).

stderr:
thread 'main' panicked at src/lsusb/names.rs:6:1:
assertion `left == right` failed
  left: None
 right: Some("Human Interface Device")
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

---- src/lsusb/names.rs - lsusb::names::protocol (line 65) stdout ----
Test executable failed (exit status: 101).

stderr:
thread 'main' panicked at src/lsusb/names.rs:6:1:
assertion `left == right` failed
  left: None
 right: Some("AT-commands (3G)")
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

---- src/lsusb/names.rs - lsusb::names::subclass (line 51) stdout ----
Test executable failed (exit status: 101).

stderr:
thread 'main' panicked at src/lsusb/names.rs:6:1:
assertion `left == right` failed
  left: None
 right: Some("Abstract (modem)")
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

failures:
    src/lsusb/names.rs - lsusb::names::class (line 37)
    src/lsusb/names.rs - lsusb::names::protocol (line 65)
    src/lsusb/names.rs - lsusb::names::subclass (line 51)

test result: FAILED. 40 passed; 3 failed; 3 ignored; 0 measured; 5 filtered out; finished in 1.26s

error: doctest failed, to rerun pass `--doc`
```

Do I need any preparation before running the tests? Any ideas about it? 🤔 

As a workaround I disabled the tests for now.

tuna-f1sh, 2024-09-26:
Hmm, I also get this failure on my main Arch install. It seems the hwdb lookup with returning nothing (not failing since it would fall back to usb-ids in that case). It's the `udev_hwdb` feature doing this. I'll have to do some digging for a fix. 

For now you can either disable tests or remove the `--all-features` flag in favour of the defaults (build and test); uses usb-ids crate rather than hwdb.bin: https://gitlab.archlinux.org/archlinux/packaging/packages/cyme/-/blob/main/PKGBUILD?ref_type=heads#L24

tuna-f1sh, 2024-09-26:
Weirdly the same call `systemd-hwdb query 'usb:v*p*d*dc03*'` on my system returns `ID_USB_CLASS_FROM_DATABASE=Human Interface Device` as an example so not sure why it's not working in code.

tuna-f1sh, 2024-09-26:
I did a quick test in https://github.com/cr8t/udev with the system hwdb.bin trying to get that path ('usb:v*p*d*dc03*') and it also fails so it appears to not be searching correctly. I suggest `--all-features` and I think I'll disable the Pure rust udev hwdb option.

orhun, 2024-09-26:
Got it, then I think removing `--all-features` flag from `cargo check` would be enough. Are these crate features documented somewhere?

tuna-f1sh, 2024-09-27:
Thanks. The hwdb one in particular is explained in the [udev section](https://github.com/tuna-f1sh/cyme?tab=readme-ov-file#linux-udev-information). I'm re-factoring the README in #26 and will probably add a section explicitly explaining each feature though.

closed 2024-09-27 by tuna-f1sh