tuna-f1sh/cyme #38: index out of bounds: the len is 10 but the index is 10

JaCzekanski, 2024-10-11:
OS: macOS Sonoma 14.6.1

Program crashes when trying to use verbose mode if I'm connected to Thunderbolt Hub.
Hub exposes audio in/out port via USB.

```
$ cyme -v
thread 'main' panicked at src/usb/descriptors/audio.rs:2668:20:
index out of bounds: the len is 10 but the index is 10
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Abort trap: 6
```

Device in question:
0x08bb 0x2912 USB audio CODEC -       12.0 Mb/s

System information:
```
usb:
        USB audio CODEC:

          Product ID: 0x2912
          Vendor ID: 0x08bb  (Texas Instruments Japan)
          Version: 1.00
          Speed: Up to 12 Mb/s
          Manufacturer: Burr-Brown from TI
          Location ID: 0x03200000 / 1
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

audio:
        USB audio CODEC:

          Manufacturer: Burr-Brown from TI
          Output Channels: 2
          Current SampleRate: 48000
          Transport: USB
          Output Source: Default

        USB audio CODEC:

          Input Channels: 1
          Manufacturer: Burr-Brown from TI
          Current SampleRate: 48000
          Transport: USB
          Input Source: Default
```

Let me know how I can get raw usb audio device descriptor for analysis.

tuna-f1sh, 2024-10-11:
Thanks. Since the line number was in the panic, I noticed what I believe is the bug - the expected length check used an incorrect value for number of channels. If you can, please can you test: https://github.com/tuna-f1sh/cyme/commit/c7c88f1c16b07d878042c9bbfac361e8bfdcb7a7

What would be useful to provide the raw descriptor information, such that I can reproduce on my computer is `cyme --tree --verbose --json > dump.json`. Would be great to have a reference since I guess you have some audio devices, which I wasn't able to fully test.

tuna-f1sh, 2024-10-11:
Actually, I realise that the `cyme --tree --verbose --json > dump.json` won't work because of the panic. If you can test [branch](https://github.com/tuna-f1sh/cyme/tree/audio-fix) and then dump (if it works) it would be useful. Thanks!

JaCzekanski, 2024-10-11:
Thanks, I'm not getting the crash anymore using `audio-fix` branch.

Here's the json dump: [dump.json](https://github.com/user-attachments/files/17344327/dump.json)

closed 2024-10-11 by JaCzekanski