tuna-f1sh/cyme #39: Linux "Can't init Global usb context, error Other"

RobertKosten, 2024-10-11:
```bash
user@untrusted:~$ RUST_BACKTRACE=full cyme
thread 'main' panicked at /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rusb-0.9.4/src/context.rs:268:28:
Can't init Global usb context, error Other
stack backtrace:
   0:     0x58c628937c37 - <unknown>
   1:     0x58c628864b4c - <unknown>
   2:     0x58c6289130d2 - <unknown>
   3:     0x58c62893caff - <unknown>
   4:     0x58c62893c8da - <unknown>
   5:     0x58c62893d11e - <unknown>
   6:     0x58c62893ccc0 - <unknown>
   7:     0x58c62893cc16 - <unknown>
   8:     0x58c62893cc03 - <unknown>
   9:     0x58c6287d48a4 - <unknown>
  10:     0x58c6287dace1 - <unknown>
  11:     0x58c62880ea11 - <unknown>
  12:     0x58c628816515 - <unknown>
  13:     0x58c6287ef6e3 - <unknown>
  14:     0x58c62882670f - <unknown>
  15:     0x7d0380d60d0a - __libc_start_main
  16:     0x58c6287ddb7a - <unknown>
  17:                0x0 - <unknown>
Aborted
```

- installed via `cargo install cyme`
- same error when executing with `sudo`
- environment is a Debian 11 VM on QubesOS

```bash
user@untrusted:~$ cyme --version
cyme 1.7.0
user@untrusted:~$ uname -a
Linux untrusted 6.1.75-1.qubes.fc32.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Feb  5 06:49:25 CET 2024 x86_64 GNU/Linux
user@untrusted:~$ cat /etc/debian_version 
11.10
```

tuna-f1sh, 2024-10-12:
This is an error from libusb. I had a search in rusb (the Rust ffi cyme uses) and only found this: https://github.com/a1ien/rusb/issues/188 - relating to a Docker container.

Only thing I would guess is the VM doesn’t have USB enabled or some configuration with USB prevents usage?

RobertKosten, 2024-10-13:
Hm, usb is enabled an available, `lsusb` doesn't error, but just shows nothing connected. When I attach a device it works with both `lsusb` and `cyme`, after detaching it *keeps* working, showing two hubs/controllers, so I'm assuming the virtual hubs aren't properly initialized until actually used and `lsusb` is more graceful about not finding anything at all?

tuna-f1sh, 2024-10-14:
Ah yes, `lsusb` probably handles it differently. The panic is rusb isn't even an unwrap in cyme, it's panicing because it has no USB context and I don't think it can be stopped at API level. I'm close to merging #26 , nusb should handle it better (no context so hopefully would just show none).

Would be great if you can test it. Builds can be found in the PR action: https://github.com/tuna-f1sh/cyme/actions/runs/11322659319 or checkout https://github.com/tuna-f1sh/cyme/tree/nusb

RobertKosten, 2024-10-14:
The Artifacts don't work for me (wrong `glibc` version), but building the `nusb` branch solves the problem and works fine, thx :-)

tuna-f1sh, 2024-10-18:
Merged `nusb` branch and released v2.0.0, which doesn't use libusb by default so closing as fixed.

closed 2024-10-18 by tuna-f1sh