tuna-f1sh/cyme #42: [feature request] man page and tab completion files

alexanderkjall, 2024-10-12:
Hi

I started to work on packaging this for Debian here: https://salsa.debian.org/rust-team/debcargo-conf/-/tree/master/src/cyme/debian

But it would be really nice to also have a man page for the project and tab completion rules.

I could write the man page myself based on the information in the README, but maybe you have some thoughts about it?

tuna-f1sh, 2024-10-12:
Thanks for helping with the packaging! There is actually a man page and shell completions generated for each release in [.doc](https://github.com/tuna-f1sh/cyme/tree/main/doc).

They are generated using the clap feature and cyme `-F=cli_generate`; `cargo run -F=cli_generate -- --gen` which calls this [function](https://github.com/tuna-f1sh/cyme/blob/a7fdf783cb25c158d75502b95236a4cafcd22e46/src/main.rs#L382). They are used in the build action to create the tarballs: https://github.com/tuna-f1sh/cyme/blob/a7fdf783cb25c158d75502b95236a4cafcd22e46/.github/workflows/build.yml#L141

Unless you were suggesting something else?

alexanderkjall, 2024-11-19:
Yes, sorry for not closing this earlier.

closed 2024-11-19 by alexanderkjall