tuna-f1sh/cyme #49: depends on non-free licensed crate

jonassmedegaard, 2024-10-23:
This project looks super cool, and I want to package it for official inclusion with Debian, but unfortunately I cannot, because
the author of the dependency crate `pci-ids` forbids inclusion in free-for-all distributions like Debian:
https://github.com/lienching/pci-ids.rs/blob/main/LICENSE

Please consider avoiding that crate, or at least make the dependency on it optional.

jonassmedegaard, 2024-10-23:
Corresponding Debian discussion is [here](bugs.debian.org/1085871).

jonassmedegaard, 2024-10-23:
The issue is now solved upstream, thanks to a swift response from the author of pci-ids at https://github.com/lienching/pci-ids.rs/issues/17

closed 2024-10-23 by jonassmedegaard

tuna-f1sh, 2024-10-23:
Perfect, thanks for both opening and solving!