tuna-f1sh/cyme #55: Seems to show an incorrect manufacturer name

zigmars, 2024-12-23:
```
> /opt/homebrew/Cellar/lsusb/1.0/bin/lsusb
Bus 001 Device 001: ID 0c68:6464 Whanam Electronics Co., Ltd. USB 2.0 Camera
```

```
> cyme --more
  1   1 1    1-1    0x0c68 0x6464 1b.02 2.00  0xef Miscellaneous 0x02 ?     0x01 Interface Association USB 2.0 Camera Sonix Technology Co., Ltd. -      480.0 Mb/s
```

as per https://www.usb.org/sites/default/files/vendor_ids100824_0.pdf
`vid = 0x0c68 == 3176` which is indeed `Whanam Electronics Co., Ltd.` (as lsusb correctly shows)

tuna-f1sh, 2024-12-23:
Hmm odd. Would you be able to `cyme -v  --json > dump.json` and attached so I can do some debugging? `cyme` will actually read the [cached device descriptor first](https://usb-ids.gowdy.us/read/UD/3176), then fallback to [usb-ids](https://usb-ids.gowdy.us/read/UD/3176) if None.

tuna-f1sh, 2024-12-23:
Also, if you `cyme --lsusb --verbose -d 0x0c68:6464` the lsusb mode output will show both the usb-ids 'idVendor' and descriptor 'iManufacturer' so we might see a difference.

zigmars, 2024-12-23:
Here is the requested dump:
[dump.json](https://github.com/user-attachments/files/18229030/dump.json)

for the `cyme --lsusb --verbose -d 0x0c68:6464`, `iManufacturer` value is the same `Sonix ...`; you can see here the output:
[cyme_lsusb_verbose_d_c68_6464.txt](https://github.com/user-attachments/files/18229057/cyme_lsusb_verbose_d_c68_6464.txt)

tuna-f1sh, 2024-12-23:
Thanks. I've pinned down what is going on here. The default `cyme` display will show the string descriptor at manufacturer index for the manufacturer block - the device is returning "Sonix..." for this. Perhaps it's a Whanam sub-brand?

`cyme --lsusb` will act like lsusb, which prints the idVendor (the static value from hwdb/usb-ids) with fallback to the descriptor string [lsusb 1](https://github.com/gregkh/usbutils/blob/master/lsusb.c#L3749), [lsusb 2](https://github.com/gregkh/usbutils/blob/3db1b647f059c9473269085739efa6716915151d/names.c#L212), [cyme equivalent](https://github.com/tuna-f1sh/cyme/blob/main/src/profiler/types.rs#L1086).

So whilst `cyme --headings --more` is different to `lsusb`, this is intended behaviour for me - I prefer to return the actual string descriptor rather than database value for the primary field.

There is a bug here however in that even with your dump, `cyme --lsusb` doesn't match `lsusb`. This is because the usb-ids lookup is returning `None`, causing a fallback to the string descriptor value. usb-ids should not be returning `None` for this since the value [appears](https://usb-ids.gowdy.us/read/UD/3176) in the database. I've tested a fork and it not [returning anything](https://github.com/woodruffw/usb-ids.rs/pull/61).

tldr; this appears to be a bug in usb-ids. I'll have a look to see if I can fix it.

tuna-f1sh, 2024-12-23:
I think I see what's going on here and it appears to be an error in usb-ids repository...

You are correct that the 0x0c68 is 3176 in decimal and the [vendor ids](https://www.usb.org/sites/default/files/vendor_ids100824_0.pdf) pdf shows decimal representation. On the usb-ids file and website however, the listings are in base16, yet it shows as the decimal! So looking up `0x3176` actually returns "Whanam Electronics Co., Ltd". Will look to get this fixed in usb-ids repository.

tuna-f1sh, 2025-02-13:
I've raised this with usb-ids but no action so far. I'm going to close this since after some more digging, this is actually re-producible with usbutils `lsusb` - since it uses the device descriptors and your device reports 'Sonix Technology Co., Ltd.'. The hwdb look-up will also fail due to above bug in usb-ids so fallback to the descriptor.

I note (based on path and my testing) that you are using the [lsusb formula](https://formulae.brew.sh/formula/lsusb) based on this [script](https://github.com/jlhonora/lsusb/blob/master/lsusb) which is actually what made me create `cyme` in the first place! It isn't actually `lsusb` but attempts to emulate with some creative bash.

Tested with the usb-gadgets and a script to emulated what your device reports: https://github.com/tuna-f1sh/usb-gadget/blob/whanam/examples/custom_interface_device.rs:

## cyme

```
❯ cyme --vidpid=0c68 --more
  3  26 2-1-1-2  3-2.1.1.2  0x0c68 0x6464 0.00  2.01  0xff VendorSpecificClass 0xff Vendor Specific Subclass 0x03 -     USB 2.0 Camera Sonix Technology Co., Ltd. serial_number 480.0 Mb/s
```

```
❯ cyme --vidpid=0c68 --lsusb
Bus 003 Device 026: ID 0c68:6464 Sonix Technology Co., Ltd. USB 2.0 Camera
```

```
❯ cyme -d 0c68: -v --lsusb

Bus 003 Device 007: ID 0c68:6464 Sonix Technology Co., Ltd. USB 2.0 Camera
Device Descriptor:
  bLength               18
  bDescriptorType        1
  bcdUSB              2.01
  bDeviceClass         255 Vendor Specific Class
  bDeviceSubClass      255 Vendor Specific Subclass
  bDeviceProtocol        3
  bMaxPacketSize0       64
  idVendor          0x0c68 [unknown]
  idProduct         0x6464 [unknown]
  bcdDevice           0.00
  iManufacturer          2 Sonix Technology Co., Ltd.
  iProduct               1 USB 2.0 Camera
  iSerialNumber          3 serial_number
```

## usbutils lsusb (linux)

```
❯ lsusb -d 0c68:
Bus 003 Device 006: ID 0c68:6464 Sonix Technology Co., Ltd. USB 2.0 Camera
❯ lsusb -V
lsusb (usbutils) 018
```

```
❯ lsusb -d 0c68: -v

Bus 003 Device 006: ID 0c68:6464 Sonix Technology Co., Ltd. USB 2.0 Camera
Negotiated speed: High Speed (480Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol         3
  bMaxPacketSize0        64
  idVendor           0x0c68 Sonix Technology Co., Ltd.
  idProduct          0x6464 USB 2.0 Camera
  bcdDevice            0.00
  iManufacturer           1 Sonix Technology Co., Ltd.
  iProduct                2 USB 2.0 Camera
  iSerial                 3 serial_number
```

(I note that the idVendor, idProduct also fallback to the descriptor here so will change in `cyme`.)

## macos script

```
❯ lsusb_script -d 0c68:
Bus 003 Device 027: ID 0c68:6464 Whanam Electronics Co., Ltd. USB 2.0 Camera  Serial: serial_number
```

So the script is actually the only one looking up the VID 0x0c68 correctly! I assume this is some database macOS uses internally since `system_profiler SPUSBDataType` returns:

```
...
USB 2.0 Camera:

  Product ID: 0x6464
  Vendor ID: 0x0c68  (Whanam Electronics Co., Ltd.)
  Version: 0.00
  Serial Number: serial_number
  Speed: Up to 480 Mb/s
  Manufacturer: Sonix Technology Co., Ltd.
  Location ID: 0x03211200 / 27
  Current Available (mA): 500
  Extra Operating Current (mA): 0
...
```

Just to prove that the usb-ids hwdb is wrong, if I change the gadget VID to 0x3176, both `cyme` and `lsusb` will return the lookup not the device descriptor manufacturer:

```
❯ cyme -d 3176: --lsusb
Bus 003 Device 008: ID 3176:6464 Whanam Electronics Co., Ltd USB 2.0 Camera
```

```
❯ lsusb -d 3176:
Bus 003 Device 008: ID 3176:6464 Whanam Electronics Co., Ltd USB 2.0 Camera
```

closed 2025-02-13 by tuna-f1sh