tuna-f1sh/cyme #56: add documentation on how to specify multiple blocks for the `--blocks <BLOCKS>` parameter

zigmars, 2024-12-23:
Tried separating with commas, did not seem to work.
In manual I could not find how the  `<BLOCKS>` must be specified, when more than one column needs to be requested.

Just now noticed - have a similar documentation problem with `--verbose` level.

tuna-f1sh, 2024-12-23:
Can answer here but will look at adding to docs - probably some example uses. For args that take a value - such as `--blocks` - one must supply the flags multiple times to chain:

`cyme --blocks bus-number --blocks icon --blocks name -b vendor-id -b product-id --blocks serial`

For args without values, one can either supply multiple times or appended:

`cyme --verbose --verbose -vv` would be verbosity 4.

zigmars, 2024-12-23:
Great, that works. Thanks for the quick reply. :)

tuna-f1sh, 2024-12-24:
Added in https://github.com/tuna-f1sh/cyme/commit/5e43cd83407d9cf1cb19ccc040fecb0ca993c977 some examples and how to chain in arg `--help`

closed 2024-12-24 by tuna-f1sh