tuna-f1sh/cyme #57: Can USB Hot Swap Detection Be Supported?

zhaofucheng, 2025-01-02:
Print Detailed Information on USB Additions and Removals

tuna-f1sh, 2025-01-02:
So some sort of history of connected devices? With an option to show new/hide hold?

zhaofucheng, 2025-01-02:
> So some sort of history of connected devices? With an option to show new/hide hold?

I want to detect USB insertion and removal events and print the details. Can you improve the support for this?

tuna-f1sh, 2025-01-03:
It feels a bit out of scope but I might look into [watch devices stream](https://docs.rs/nusb/latest/nusb/fn.watch_devices.html). 

I could see a 'watch' option with a dynamic tree, which pops/pushes devices as they connect. What sort of detailed information on connect/disconnect are you thinking? Is it to have a terminal open showing this for a human or piped into something? The later I would suggest implementing something else using the nusb function linked above.

zhaofucheng, 2025-01-03:
> It feels a bit out of scope but I might look into [watch devices stream](https://docs.rs/nusb/latest/nusb/fn.watch_devices.html).
>
> I could see a 'watch' option with a dynamic tree, which pops/pushes [...]

This is what I need, but it is not implemented in your code

tuna-f1sh, 2025-01-29:
I've made a proof of concept for this if you're interested: #58

tuna-f1sh, 2025-04-05:
Merged #58

closed 2025-04-05 by tuna-f1sh