tuna-f1sh/cyme #65: cyme watch No UI interface is required.

zhaofucheng, 2025-04-25:
Only after execution, when a USB device is inserted, display the information of the currently inserted USB device. When it is unplugged, display the unplugged device information, just like a log. Currently, it is also possible to filter and monitor specified USB devices. In the current implementation you've done, one can only judge the connection by looking at the "E" and "O" icons. What I want is to use Python to call your cyme.exe to parse the detailed information of insertion and removal, similar to the pyudev library on Linux. I will implement this on Windows and abandon the WMI detection. The information detected by WMI is incomplete, and it is not case-sensitive. The cyme you've implemented is exactly what I need. Thank you.

tuna-f1sh, 2025-04-25:
I guess you mean a live updating `--json` output.

zhaofucheng, 2025-04-25:
> I guess you mean a live updating `--json` output.

Yes, you're right. There's no need to display all devices. Just update the currently inserted or disconnected device in the JSON. I only need to detect the update of the JSON file.

tuna-f1sh, 2025-04-28:
I added functionality for `--json` to `cyme watch` in commit above. One can filter with `jq` for example: `cargo run -- --json watch | jq -n '.. | .last_event?'`. I feel like an IPC might be nicer for this but maybe this is good enough.

Try it out and see if it works for your use case.

PR #66 (2025-05-01): feat: --json stream for watch #65
Adds `--json` output functionality to `watch` subcommand. Streams '\n' separated json dump of system USB profiling with each hot plug event.

I feel like a IPC protocol would be better for sharing between processes but this is a easy addition for using cyme within code.

Some `jq` filtering examples:

## Just show name and event

`cargo run -- --json watch | jq '.buses[] | .devices[]? | {name, last_event}'`

## Filter only hot-plug events and dump changed device

```
jq '
  .buses[] | .devices[]?
  | select(
      (.last_event | has("Connected")) or
      (.la [... cut]