tuna-f1sh/cyme #67: Thank you for your design efforts. The watch function is really great.

zhaofucheng, 2025-05-08:
watch--json. It seems that the data accumulates every time for connection and disconnection when obtaining json. Is this a bug? Moreover, there are too many levels in the json data, and it's not obvious to obtain connection and disconnection information. Is there a better design plan? 

import asyncio
import subprocess
import json
def on_connected(buses):
    print(json.dumps(buses, indent=4))

def on_disconnected(buses):
    print(json.dumps(buses, indent=4))
async def watch_cyme():
    process = await asyncio.create_subprocess_shell(
        'cyme --json watch',
        stdout=subprocess.PIPE,
        stderr=subprocess.PIPE
    )

    try:
        while True:
            line = await process.stdout.readline()
            if not line:
                continue
            try:
                data = json.loads(line)
                buses = data['buses'][0]
                devices=buses['devices']
                last_event=devices[-1]['last_event']
                if 'connected' in last_event:
                    on_connected(buses)
                elif 'disconnected' in last_event:
                    on_disconnected(buses)

            except json.JSONDecodeError:
                print(f"Failed to decode JSON: {line.strip()}")

    except asyncio.CancelledError:
        print("Watcher task cancelled")
    finally:
        process.kill()
        await process.wait()
asyncio.run(watch_cyme())

zhaofucheng, 2025-05-08:
For connection and disconnection, there often isn't an obvious name. Most of the time it works fine, but in some individual cases, it keeps printing like this: 'last_event': {'profiled': '2025-05-08T12:05:03.773021400+08:00'}}]   profiled

tuna-f1sh, 2025-05-08:
Thanks. I did mention in https://github.com/tuna-f1sh/cyme/pull/66 it's non-ideal way for process to process communication - a specific IPC protocol with protobuf or something would be nicer but it's a bit out of scope for me.

Yes the json dump is repeated each update, including non-changed devices. I think this is your question regarding accumulation? It's the MVP for me for this json stream. 

One should be able to message the json into what you are trying to do. In your case you probably want to pass the device json matching `'last_event': 'connected' || 'disconnected'` (pseudo code) to your callbacks.

zhaofucheng, 2025-05-08:
Instead of being 'connected' || 'disconnected', its status is 'profiled'. In most cases, it can recognize 'connected' || 'disconnected', but in some individual cases when plugging or unplugging, the status is always 'profiled'.

tuna-f1sh, 2025-05-08:
What platform are you testing on?

zhaofucheng, 2025-05-08:
win

tuna-f1sh, 2025-05-09:
Ah ok. I think I know what this is and it's a bug due to how windows provides Bus numbers. 

I'll have to rethink how find reconnected device locations on Windows.

tuna-f1sh, 2025-05-09:
I updated with a fix. If you're able to test pre release that would be great. There are compiled artefacts here: https://github.com/tuna-f1sh/cyme/actions/runs/14929694325#artifacts

zhaofucheng, 2025-05-10:
Now the connection event “connected” can be detected correctly, but the disconnection event still shows as “profiled” instead of “disconnected”.

closed 2025-07-29 by tuna-f1sh