tuna-f1sh/cyme #71: Show default output block options in help output

fidgetingbits, 2025-06-12:
It would be useful to show whatever the default `--block` values are for cyme output in the `--help` output somewhere. This way if you want to tweak something you can work from the existing default output layout as a starting point. 

For example, I want to tweak the default output to not include the serial at all (not just mask/obfuscate), but in order to do that I have to include a complete set of all `--block` options to match the existing output, and I don't easily know what was used, so it's a bit of extra work to manually reconstruct. If the default set of options was easily accessible somewhere in the output, it would make tweaking output layouts easier/faster.

If this already exists somewhere that I missed, please let me know.

tuna-f1sh, 2025-06-12:
The generated example config includes the default args, so one way to achieve this is to copy to $XDG_CONFIG_HOME/cyme/cyme.json and remove the serial block: https://github.com/tuna-f1sh/cyme/blob/main/doc/cyme_example_config.json

I agree though that the block args in help would be a helpful feature as I've found myself wanting to do the same. Potentially even an exclude flag would be helpful - then one could add and remove keeping default blocks more easily. I'll have a think.

fidgetingbits, 2025-06-13:
I just tried it out and works nicely. Thanks for that.

One thing that came to mind just playing around, which is maybe suited for a follow-up issue, is that it would be a lot easier for the user to just comma-delimit blocks in a single `--blocks` so like `--blocks vid-pid,bus-power`. This would make the default in the `--help` output a lot cleaner/smaller too.

As for add and removing some I agree as it stands with the `--block-operation` and multiple `--blocks` it would be a bit too overwhelming. But, you could consider extending the comma-delimited `--blocks` idea above and have `+` and `-` prefix for each block in the list to indicate what you want to add/remove for instance `--blocks -serial,+vid-pid` or something.

tuna-f1sh, 2025-06-13:
The csv option is actually a derive from clap so was easy to add 🙂 - another nice QoL improver! https://github.com/tuna-f1sh/cyme/pull/72/commits/afbe7f153b45a77e1730d6ea5d0910d9c0b580b4#diff-42cb6807ad74b3e201c5a7ca98b911c5fa08380e942be6e4ac5807f8377f87fc

closed 2025-06-13 by tuna-f1sh